sayanee/belle #6: interchange the bell connectors

sayanee, 2017-03-20:
> I'm a little confused. When you press the button. It pulls the line low. Which means, after the Schmidt inverter, it's going to be high. Generally, reset pins are active​ low. I'm not sure about the esp though. And Your speaker logic is inverted. When the button is pressed, the output of the Schmidt trigger is high, so the speaker sees 3.3V at both pins. So it won't sound. And it will always beep when not pressed

shannietron, 2017-03-20:
If you are going to continue sinking current through the inverter, it'll be way easier to just ground the other pin of the buzzer.

sayanee, 2017-03-21:
Related to #8

closed 2017-05-01 by sayanee

sayanee, 2017-07-10:
From [ESP8266 System Description](https://espressif.com/en/support/download/documents) Version 2.2, 2017

> Pin32 serves as an RST pin which can be left floating when it is not used. The reset pin is held low when the chip is enabled. In order to avoid resets caused by external interference, the lead is generally required to be short, and no external pull-up resistor is necessary.