sayanee/belle #14: Unable to flash with ESP12F modules

sayanee, 2017-04-22:
## Steps to reproduce

1. Wire up an `ESP12F module`
    - `GND` to `0V`
    - `GPIO15` to `0V`
    - `GPIO2` to `3V3
    - `GPIO0` to `0V`
    - `GPIORX` to `TX` on serial
    - `GPIOTX` to `RX` on serial
    - `VCC` to `3V3`
    - `CHPD` to `3V3`
    - `RST` the chip by connecting rest to `0V` briefly before flashing
    [image]
1. Connect to the MacBook via the [serial console cable](https://www.adafruit.com/product/954)
1. Open up Arduino IDE to flash a blinky program

    ```c
    void setup() {
      pinMode(LED_BUILTIN, OUTPUT);
    }

    void loop() {
      digitalWrite(LED_BUILTIN, LOW);                   
      delay(1000);                   
      digitalWrite(LED_BUILTIN, HIGH);
      delay(2000);
    }
    ```
    
    [image]

## Actual result

Arduino is having an error when flashing

[image]

## Expected result

Arduino should be successful in flashing the blinky progam

sayanee, 2017-04-22:
@shannietron any idea? 🤔

notthetup, 2017-04-22:
Instructions from Adafruit for flashing firmware.

Maybe GPIO0 can't be tied to ground all the time. https://learn.adafruit.com/adafruit-huzzah-esp8266-breakout/using-arduino-ide

> Hold down the GPIO0 button, the red LED will be lit
> While holding down GPIO0, click the RESET button
> Release RESET, then release GPIO0
> When you release the RESET button, the red LED will be lit dimly, this means its ready to bootload

sayanee, 2017-04-22:
> turn on serial monitor then reset, can you see something like 'waiting for download'?

nope, nothing

> that's not right. even if not enter program made there need be something. maybe check serial connection?

sayanee, 2017-04-25:
Wired it like Huzzah and the serial monitor gives:

```sh
Fatal exception (0): epc1=0x40100003, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
```

sayanee, 2017-05-01:
# Solution

1. Use a proper [FTDI chip](https://world.taobao.com/item/13702527783.htm?fromSite=main&ut_sk=1.UpdigsCXw0cDAECxW5pO7mwr_21380790_1493626825951.Copy.1&sourceType=item&price=25&suid=D6574792-6BAA-4E74-A821-5A4BCA83B2A8&cpp=1&shareurl=true&spm=a313p.22.2r6.36666827659&short_name=h.gE7eOv&cv=2jSFZGYevmU&sm=d06db6&app=chrome)
1. Be careful of shorting ESP12 module pins while soldering

sayanee, 2017-05-02:
## Normal mode

> serial monitor output

[image]

## Boot mode

> serial monitor output

[image]

notthetup, 2017-06-29:
https://github.com/esp8266/esp8266-wiki/wiki/Boot-Process#esp-boot-modes

sayanee, 2017-06-30:
| GPIO15 | GPIO0 | GPIO2 | Mode | Description | Binary | Decimal
| ------ | ------ | ------ | ------ | ------ | ------ | ------
| L | L | H | UART | Download code from UART | 001 | 1
| L | H | H | Flash | Boot from SPI Flash | 011 | 3
| H | x | x | SDIO | Boot from SD-card | 100, 101, 110, 111 | 4, 5, 6, 7

sayanee, 2017-07-01:
See [edited solution](https://github.com/sayanee/belle/issues/14#issuecomment-298308491)

closed 2017-07-01 by sayanee