sayanee/belle #18: DFM tweaks

kennethlimcp, 2017-05-09:
1. R3 can be rotated 90 degrees and placed between M3 mounting hole and buzzer.

  This would aid soldering of ESP-12 pads and makes any reworking easier.

2. Board information can be moved upwards to avoid last "e" in `sayanee` from overlapping with via

3. M3 label can be placed nearer to NTPH holes

4. Via is too closed to bottom trace (blue arrow)

[image]

5. There seems to be an issue with the top copper pour set to "+3v3"? 

Usually the planes are tied to ground instead of a voltage, unless you are working with >2 layer boards where a layer is required to connect components to 3v3/5v.

If this is switched, there will be lesser vias with R4, C1, R1, C4, C7, C3, C6, C2 all having one side of the pad connected to the ground plane directly.

sayanee, 2017-05-10:
- [ ] move up board information
- [ ] place M3 label nearer to NPTH
- [ ] move via near the LDO

sayanee, 2017-05-10:
@kennethlimcp 

> There seems to be an issue with the top copper pour set to "+3v3"

In the current design, the top layer is `3V3 and bottom layer is `GND`. If I switch them around, there seems to be around an equal number of vias for `3V3` instead of `GND`.

closed 2017-05-10 by sayanee